frc3512/Robot-2020 #44: Implement leading/lagging the target while the drivetrain is moving

KyleQ1, 2020-06-02:
* Our current solution doesn't lead/lag because it assumes instantaneous time of flight. We need to include the ball's actual velocity vector out of the shooter and the drivetrain velocity vector to find the desired turret angle relative to the chassis.

* Here's an article on the math we need with some messy code: https://www.codeproject.com/articles/990452/interception-of-two-moving-objects-in-d-space

willji0023, 2020-06-19:
This is what I got. Thoughts?
https://docs.google.com/drawings/d/18WzkPOciQZ1gUmpbT0Rz63n-URB6hmX9EIIBCdBfPaA/edit?usp=sharing

calcmogul, 2020-06-19:
You should explain your setup (lay out your definitions like the coordinate frame, what vectors are involved, etc.) and show more of your work. It isn't clear from the diagram alone that the equation you got at the end is correct, so I'm having trouble verifying it.

commit e65f0b40 (2020-09-24): Add aiming lead/lag on moving targets

Closes #44.
commit 56329bad (2020-09-24): Add aiming lead/lag on moving targets

Closes #44.
commit 125f40d4 (2020-09-24): Add aiming lead/lag on moving targets

Closes #44.
commit c083a633 (2020-09-29): Add aiming lead/lag on moving targets

Closes #44.
closed 2020-10-29 by calcmogul