frc3512/Robot-2020 #64: Make a way to stream controller data live to a python plotting script

calcmogul, 2020-09-05:
The current CSV approach is hard to manage remotely for testing/tuning purposes because it isn't clear which file contains the desired data. Furthermore, the CSVs don't reliably write to the roboRIO's flash.

This is basically what https://github.com/frc3512/LiveGrapher does, but I think we can do better now from a usability perspective.

KyleQ1, 2020-09-08:
How is it going to be different from LiveGrapher?

calcmogul, 2020-09-09:
We should make a wrapper around the LiveGrapher host library (the part that lives on the robot) to automatically plot the controller data. We can make a modified version of ControllerLogger that uses LiveGrapher as the backend instead of .csv files.

commit 15836b62 (2020-09-11): Add LiveGrapher backend to ControllerLogger

Fixes #64.
commit d737e537 (2020-09-11): Add LiveGrapher backend to ControllerLogger

Fixes #64.
commit 3551431f (2020-09-11): Add LiveGrapher backend to ControllerLogger

Fixes #64.
commit 3c54e993 (2020-09-11): Add LiveGrapher backend to ControllerLogger

Fixes #64.
closed 2020-09-19 by calcmogul